troystribling/BlueCap #7: "CenteralError.peripheralNotConnected" Does Not Exist

jcady92, 2015-07-14:
Here's an excerpt from the simple Central example:

```
let peripheralDiscoveredFuture = peripheralConnectFuture.flatmap {(peripheral, connectionEvent) -> Future<Peripheral> in
    if peripheral.state == .Connected {
        return peripheral.discoverPeripheralServices([serviceUUID])
    } else {
        let promise = Promise<Peripheral>()
        promise.failure(CenteralError.peripheralNotConnected)
        return promise.future
    }
}
```

Two things:
1. `CenteralError` should be `CentralError`
2. `peripheralNotConnected` is not a member of the CentralError enum.  There is only one member, `IsScanning`.

jcady92, 2015-07-14:
The enum doesn't work with that anyway, given promise.failure accepts a NSError as an argument.

Maybe you meant `BCError.peripheralDisconnected`?

troystribling, 2015-07-28:
The example was taken from the Central example here https://github.com/troystribling/BlueCap/blob/master/Examples/Central/Central/ViewController.swift#L13-28. 

CenteralError.peripheralNotConnected is an application defined error. I will update the example to include the definition so that there is not any confusion.

closed 2015-07-28 by troystribling