troystribling/BlueCap #10: support for cocoa pods

magellan314, 2015-10-18:
More of a request rather than an issue. Do you intend to add support for cocoa pods for BlueCap? It makes updates and new features / bug fixes much easier to incorporate than manually adding to projects.

troystribling, 2015-10-19:
I will look into it. When I started Cocoa Pods did not support swift. Will update issue when done.

rph8, 2015-11-03:
This is definitely a great idea. It makes updating BlueCap much easier, because one only has to do a "pod install". CocoaPods has also supported Swift for some time now.

troystribling, 2016-02-29:
I am close to starting this. Possibly in a week or two.

bshewmak, 2016-05-02:
I realize these comments are month's old, but is there still any chance that you are still going to add the cocoa pods support? As said previously, it makes it that much easier to implement into a current project.

troystribling, 2016-05-02:
I intend to add cocoa pods support. I am currently finishing up a review of test cases. After that I will add cocoa pods support.

troystribling, 2016-06-04:
Just release version 0.1.0 pod. Carthage is also supported. Still need to update documentation.

closed 2016-06-04 by troystribling