troystribling/BlueCap #11: not able to subscribe to characteristic with indicate property

TorstenRobitzki, 2015-10-20:
Hi,
first of all, let me thank you for this great tool! I currently use BlueCap to test the implementation of a characteristic that has the properties Write and Indicate. I would expect that I'm able to enable Notifications for that characteristic, but I'm not. Is this a know limitation or bug? Is there an other way to write the indication bit to the client characteristic configuration descriptor?

kind regards,
Torsten

troystribling, 2015-10-21:
Support for indications has not yet been implemented. I will look into adding it as a feature. This issue can be left open until it is completed.

TorstenRobitzki, 2015-10-21:
That would be really cool! Thank you.

troystribling, 2015-10-24:
I just committed the changes needed to support indications in the BlueCap app and BlueCapKit. An example Peripheral app sending indications if here https://github.com/troystribling/BlueCap/tree/master/Examples/PeripheralWithIndication.

closed 2015-10-24 by troystribling