troystribling/BlueCap #12: Support For Writes Without Responses

jcady92, 2015-10-30:
From this: https://github.com/troystribling/BlueCap/blob/master/BlueCapKit/Central/Characteristic.swift#L321

It looks like BlueCap is hardcoded to always write with response.  I'd like to be able to write without response for my application.

There are some complications implementing write without responses and sending a large amount of data.  Since there are no callbacks to let you know when the write is done, the writes have to be spaced out based on the CI.  I don't know of a good way to do this right now, I actually just asked this question on SO (http://stackoverflow.com/questions/33442472/ios-bluetooth-le-proper-way-to-write-without-response).

It'd be nice to be able to just pass a large chunk of data in and have the library handle the timing, but since that's not readily defined, the best way to do it for this library is probably just to enable the functionality of WriteWithoutResponse which just passes the data straight into the writeValue method and leaves the timing/write frequency up to the user's application.

troystribling, 2015-11-01:
I added support for CBCharacteristicWriteType in all write methods to the refactor branch. I will close this issue when I merge to master.

jcady92, 2015-11-02:
Sounds good, looking forward to pulling it.

troystribling, 2015-11-06:
Write requests with type:CBCharacteristicWriteType have been merged with master.

closed 2015-11-06 by troystribling