troystribling/BlueCap #13: Notifications Fail To Properly Deliver Packets

jcady92, 2015-10-31:
I'm using BlueCap as a central to connect to a peripheral that has a notification characteristic.  I subscribe to that characteristic and then set up notifications like so:
```
//Subscribe to the notification
if let service = peripheral.service(SERVICE_UUID), txCharacteristic = service.characteristic(TX_CHARACTERISTIC_UUID)
{
    //Set subscription
    txCharacteristic.setNotifyValue(true)
    
    ///Set up updates for the notification
    let updatesFuture = txCharacteristic.recieveNotificationUpdates()
    
    updatesFuture.onSuccess({
        characteristic in
        
        if characteristic.dataValue != nil
        {
            var data = [UInt8](count: characteristic.dataValue!.length, repeatedValue: 0)
            characteristic.dataValue!.getBytes(&data, length: data.count)
            
            var byteStr = ""
            for byte in data
            {
                byteStr += String(format: "%02X ", arguments: [byte])
            }
            print("updatesFuture: \(byteStr): \(String(UTF8String: dispatch_queue_get_label(DISPATCH_CURRENT_QUEUE_LABEL))!)")
        }
    })
    
    updatesFuture.onFailure({
        error in
        
        //Error
    })
}
else
{
    //Error
}
```

I also added a log similar to the one above here: https://github.com/troystribling/BlueCap/blob/master/BlueCapKit/Central/Peripheral.swift#L529

My peripheral sends more than 20 bytes of data across a set of packets.  In this case, the data it was attempting to send was 5 packets, 72 bytes.

Here are the logs from the first attempt:
```
1a. didUpdateValueForCharacteristic: 7E 01 53 00 00 00 01 08 06 12 0C 50 94 BF 57 86 12 DA DD FC : com.gnos.us.central.main
2a. didUpdateValueForCharacteristic: 92 93 F9 1A 40 18 15 8A 78 D8 D3 1C 3A 66 30 E2 DC CB 66 53 : com.gnos.us.central.main
1b. updatesFuture:                   7E 01 53 00 00 00 01 08 06 12 0C 50 94 BF 57 86 12 DA DD FC : com.apple.main-thread
3a. didUpdateValueForCharacteristic: D4 BE 03 7A E3 AC 57 49 CE 62 C4 BD 01 82 E9 1D 2E 39 07 8F : com.gnos.us.central.main
2b. updatesFuture:                   D4 BE 03 7A E3 AC 57 49 CE 62 C4 BD 01 82 E9 1D 2E 39 07 8F : com.apple.main-thread
4a. didUpdateValueForCharacteristic: 29 00 2F A1 25 F2 AF AB C1 17 BE AC 9D C7 92 BD 65 7C 67 B1 : com.gnos.us.central.main
3b. updatesFuture:                   D4 BE 03 7A E3 AC 57 49 CE 62 C4 BD 01 82 E9 1D 2E 39 07 8F : com.apple.main-thread
4b. updatesFuture:                   29 00 2F A1 25 F2 AF AB C1 17 BE AC 9D C7 92 BD 65 7C 67 B1 : com.apple.main-thread
5a. didUpdateValueForCharacteristic: B3 B6 5A DA 91 B0 93 19 7A AF 3D 7F : com.gnos.us.central.main
5b. updatesFuture:                   B3 B6 5A DA 91 B0 93 19 7A AF 3D 7F : com.apple.main-thread
```

The data arrives correctly in the proper order, as logged by the didUpdateValueForCharacteristic logs.  But when it is delivered to the future (as logged by the updateFuture logs), there is a packet missing.  Packet 2a was never delivered and packet 3a was delivered twice (2b and 3b).  

Here's another example of a retry of the same set of data:
```
1a. didUpdateValueForCharacteristic: 7E 01 53 00 00 00 01 08 06 12 0C 50 94 BF 57 86 12 DA DD FC : com.gnos.us.central.main
2a. didUpdateValueForCharacteristic: 92 93 F9 1A 40 18 15 8A 78 D8 D3 1C 3A 66 30 E2 DC CB 66 53 : com.gnos.us.central.main
3a. didUpdateValueForCharacteristic: D4 BE 03 7A E3 AC 57 49 CE 62 C4 BD 01 82 E9 1D 2E 39 07 8F : com.gnos.us.central.main
1b. updatesFuture:                   7E 01 53 00 00 00 01 08 06 12 0C 50 94 BF 57 86 12 DA DD FC : com.apple.main-thread
2b. updatesFuture:                   D4 BE 03 7A E3 AC 57 49 CE 62 C4 BD 01 82 E9 1D 2E 39 07 8F : com.apple.main-thread
3b. updatesFuture:                   D4 BE 03 7A E3 AC 57 49 CE 62 C4 BD 01 82 E9 1D 2E 39 07 8F : com.apple.main-thread
4a. didUpdateValueForCharacteristic: 29 00 2F A1 25 F2 AF AB C1 17 BE AC 9D C7 92 BD 65 7C 67 B1 : com.gnos.us.central.main
5a. didUpdateValueForCharacteristic: B3 B6 5A DA 91 B0 93 19 7A AF 3D 7F : com.gnos.us.central.main
4b. updatesFuture:                   29 00 2F A1 25 F2 AF AB C1 17 BE AC 9D C7 92 BD 65 7C 67 B1 : com.apple.main-thread
5b. updatesFuture:                   B3 B6 5A DA 91 B0 93 19 7A AF 3D 7F : com.apple.main-thread
```

I suspect there is some foul play going on with how the futures manage firing data in the queues.  It looks like sequential promises being delivered to a futurestream are not guaranteed to be delivered back to the result in the same order.

I'll admit I'm not well versed in futures, so it's very possible I'm just not doing something right.  If that's the case, would you mind advising the proper way to get notifications delivered in the order in which they're received?

troystribling, 2015-11-01:
I am currently doing a refactor to facilitate testing. With the changes I will be able to write unit tests that have a more natural flow through the code. I will add a test case for this scenario which currently does not have a test case.

jcady92, 2015-11-02:
Sounds good, thanks Troy!

troystribling, 2016-02-15:
The problem was cause by  not persisting the value of the notification when received. Now the notification value is pasted in he FutureStream completion.

closed 2016-02-15 by troystribling