troystribling/BlueCap #14: Question about multiple writes in sequence

arvindraghavan, 2015-11-02:
I was reviewing the code for characteristic writes, and it looks like the code may be fragile for multiple writes in quick succession (specifically when `writeValue` is called multiple times before the `didWriteValueForCharacteristic` is called back for the first write). It looks like the `writing` flag in `CharacteristicImpl` can be set to `false` even if a write may be pending. Mock example timeline:

1. First `writeData()` call 
2. Second `writeData()` call
3. First `writeData` completes succesfully, `didWrite()` is called, sets `writing` flag to `false`
4. Second `writeData` fails, and times out. Since the `writing` flag is now already `false`, `timeoutWrite` will NOT call `writePromise.failure()`

Did I miss something? It looks like iOS internally handles sequencing of multiple writes, but doesn't provide a way for the client to know which write succeeded. See for example: 

http://stackoverflow.com/questions/10977649/writing-multiple-bluetooth-low-energy-characteristic-values-with-response

Therefore, it looks like a CoreBluetooth wrapper will have to queue up write requests so that it can echo the data written in the response-- a feature that the iOS stack currently lacks. 

I'm still a relative newbie, so am I missing something here? Thanks for sharing your code, and any pointers are appreciated.

troystribling, 2015-11-03:
Thanks, I will take a look.

troystribling, 2016-02-15:
Writes are queued now. Thanks.

closed 2016-02-15 by troystribling