troystribling/BlueCap #15: Writing to characteristic not supported, but why?

rph8, 2015-11-03:
I get the error mesage "Error Domain=BlueCap Code=5 "Characteristic write not supported" UserInfo={NSLocalizedDescription=Characteristic write not supported}" in the log and I don't get why.
I am not using a profile, but followed the Central (without profile) example and it seems like the writing property is not enabled.
But why is that?
How do I enable the property?
I tested this with pure CoreBluetooth before and writing to the same characteristic was not an issue.
I am talking to a UART emulation over BLE service using a Nordic nRF8001 and the library by RedBearLabs -> https://github.com/RedBearLab/nRF8001/.

```
// on power, start scanning. when peripheral is discovered connect and stop scanning
    let manager = CentralManager.sharedInstance
    let peripheralConnectFuture = manager.powerOn().flatmap {_ -> FutureStream<Peripheral> in
      manager.startScanningForServiceUUIDs([self.serviceUUID], capacity:10)
      }.flatmap {peripheral -> FutureStream<(Peripheral, ConnectionEvent)> in
        manager.stopScanning()
        return peripheral.connect(10, timeoutRetries:5, disconnectRetries:5)
    }
    
    // on .Timeout and .Disconnect try to reconnect on .Giveup terminate connection
    peripheralConnectFuture.onSuccess{(peripheral, connectionEvent) in
      switch connectionEvent {
      case .Connect:
        break
        // TODO
      case .Timeout:
        peripheral.reconnect()
      case .Disconnect:
        peripheral.reconnect()
      case .ForceDisconnect:
        break
        // TODO
      case .Failed:
        break
        // TODO
      case .GiveUp:
        peripheral.terminate()
      }
    }
    
    // discover services *and* characteristics
    let characteristicsDiscoveredFuture = peripheralConnectFuture.flatmap {(peripheral, connectionEvent) -> Future<Peripheral> in
      if peripheral.state == .Connected {
        return peripheral.discoverPeripheralServices([self.serviceUUID])
      } else {
        let promise = Promise<Peripheral>()
        promise.failure(CentralError.peripheralNotConnected)
        return promise.future
      }
    }
    
    let writeCharacteristicFuture = characteristicsDiscoveredFuture.flatmap {peripheral -> Future<Characteristic> in
      if let service = peripheral.service(self.serviceUUID), characteristic = service.characteristic(CBUUID(string:self.rxUUID)) {
        let message = WhisperParameterlessMessage(rawValue: WhisperMessageType.Nop.rawValue)
        return characteristic.write(message!, timeout:20.0)
      } else {
        let promise = Promise<Characteristic>()
        promise.failure(CentralError.rxCharacteristicsNotFound)
        return promise.future
      }
    }
    writeCharacteristicFuture.onSuccess {characteristic in
     // Log
    }
    writeCharacteristicFuture.onFailure {error in
      // Log
    }
    
    let writeCharacteristicFuture2 = writeCharacteristicFuture.flatmap { characteristic -> Future<Characteristic> in
     // Log
      let message = WhisperInteractiveChannelSetMessage(rawValue1: [0], rawValue2: self.kDefaultMessageParameters)
      return characteristic.write(message!, timeout:20.0)
    }
    
    writeCharacteristicFuture2.onSuccess {characteristic in
      // Log
    }
    writeCharacteristicFuture2.onFailure {error in
      //  Log
    }
```

troystribling, 2015-11-04:
For a central to write a characteristic it must have either CBCharacteristicProperties.Write or CBCharacteristicProperties.WriteWithoutResponse enables. Is this the case for your application? See
 https://github.com/troystribling/BlueCap/blob/refactor/BlueCapKit/Central/Characteristic.swift#L68-L70

rph8, 2015-11-04:
Yes, but that's not quite what I meant.
The line of code you reference only reads these properties.
But can I actually set them?
So I guess my real question here is: Are CBCharacteristicProperties read only and if not how do I set them?
When I tried with CoreBluetooth trivially, I did not enable those properties, they must have apparently been enabled automatically and now they are not.

troystribling, 2015-11-04:
For Characteristic, which wraps CBCharacteristics, properties are read only. https://developer.apple.com/library/prerelease/ios/documentation/CoreBluetooth/Reference/CBCharacteristic_Class/index.html#//apple_ref/occ/instp/CBCharacteristic/properties. You can write the properties for a MutableCharacteric which wraps CBMutableCharacteristic in the initializer. https://developer.apple.com/library//mac/documentation/CoreBluetooth/Reference/CBMutableCharacteristic_Class/index.html. You would only use a MutableCharacteristic in PeripheralManager applications not CentralManager and you are using a CentralManager.

I have written applications using the RedBear Labs shield for Arduino. You have to set the characteristic as writable in the GATT profile definition on the Arduino. You cannot change the properties on the iPhone.

rph8, 2015-11-04:
Yeah, but I must be doing something wrong with BlueCapKit I guess or there is a bug. So your explanation is unfortunately not really helpful since I am not running a peripheral on the iOS end, but a central. I just tested it again with plain CoreBluetooth and the exact same shield. I can perfectly write to the exact characteristic whereas with BlueCapKit I get the error above stating that same characteristic that plain CoreBluetooth writes has writing not supported when using BlueCapKit.

This works with CoreBluetooth, but the equivalent in BlueCapKit above gives me said error:
```
var messageBytes: [UInt8] = [MessageType.ChannelSet.rawValue, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
    peripheral!.writeValue(NSData(bytes: &messageBytes, length: 13), forCharacteristic: txCharCB!, type: .WithoutResponse)
   LogDebug(self, format: "Did write ChannelSet message to RX.")
´´´

troystribling, 2015-11-04:
Is writing enabled on the peripheral? It is failing because blucap thinks
the characteristic does not think it is writable. Could you check the
characteristic properties and add them to the issue?

closed 2016-02-04 by troystribling