troystribling/BlueCap #23: Futures: `onFailure` blocks do not prevent failures from falling through to later `onFailure` blocks

cbowns, 2016-02-03:
I've got a question about BlueCap (or perhaps it's for SimpleFutures) and its handling of `onFailure`:

I've got a bunch of simple BT code strung together to discover, filter, and connect to a device. I've got `onSuccess` and `onFailure` blocks written for each of my calls.

Specifically, I'm scanning for all nearby devices, and using a `withFilter:` block to search for a peripheral with a specific name pattern. When I reject the devices by returning `false`, the `onFailure` block logs, "Filter failed", as I expected.

```
        peripheralConnectFuture.onFailure { (error) -> Void in
            print("Error connecting to the device. \(error.localizedDescription)")
        }
```

However! I'm seeing something frustrating in subsequent futures:

After my filter passes, I connect to the peripheral. I discover services in `peripheralConnectFuture`:

```
let peripheralServicesDiscoveredFuture = peripheralConnectFuture.withFilter { (peripheral, connectionEvent) -> Bool in
…
```

In its `onFailure` block,

```
         peripheralServicesDiscoveredFuture.onFailure { (error) -> Void in
            print("Error discovering services: \(error.localizedDescription)")
        }
```

I'm receiving failure callbacks from the `withFilter` block, for devices I rejected.

Is this by design, or is this a bug?

cbowns, 2016-02-03:
After reading the [SimpleFutures documentation](https://github.com/troystribling/SimpleFutures#map), it appears this is by design. Cool.

So… if I wanted my `onFailure` calls to only be called for failures of their immediately-associated future, how can I do that? It seems like I'd need to modify SimpleFutures to do so… ?

troystribling, 2016-02-04:
Let me be sure I understand. You have a chain of Futures and FutureStreams and you are getting failures in peripheralConnectFuture that are not failures in peripheralServicesDiscoveredFuture in the peripheralServicesDiscoveredFuture onFailure callback?

cbowns, 2016-02-04:
I think we're saying the same thing, but lemme rephrase:

I have Future1, which runs through a `withFilter` block, and has `onSuccess` and `onFailure` blocks. Future2 is `flatMapped` from Future1, and also has `onSuccess` and `onFailure` blocks.

When my code in `withFilter` block fails a received event, the `onFailure` blocks for Future1 *and* Future2 are both called, and both receive the same "Filter failed" error.

cbowns, 2016-02-04:
I was looking at it more last night (I added a stringly-typed `error.localizedDescription` check in my `onFailure` blocks after reading through SimpleFutures a bit more). I'll open up SimpleFutures a bit and see how to make this change.

closed 2016-02-04 by cbowns

troystribling, 2016-02-04:
Cool. We are talking about the same thing. Take a shot at change. I will
think about it as well.