troystribling/BlueCap #25: Return richer error objects

cbowns, 2016-02-17:
As a boomerang/follow-up to https://github.com/troystribling/SimpleFutures/issues/1:

BlueCap should return a richer error object type than a simple NSError. This richer object should contain more information about the failure's origin: was it a failed filter? propagation of an upstream future's failure? etc etc.

cbowns, 2016-02-17:
I'd be happy to stub out an example of what's useful on the client side of this, and try to integrate it into BlueCap.

troystribling, 2016-02-17:
Sure, you can make change when finished. The error handling in the app and examples is simple. It should be a straight replacement of NSError. I think the SFError class should be a child of NSError. Would something as simple as a tag indicating where it should be handled enough. Each error handler could check for its tag. Tag could be Int then you could use enums to represent where processing should occur. First, error handler could act as router and add tags if it does not process errors. I think this will be easier than rewriting error propagation.

cbowns, 2017-02-21:
I won't be getting to this anytime soon.

closed 2017-02-21 by cbowns