troystribling/BlueCap #27: Swift 2.2 deprecation warnings

cbowns, 2016-02-25:
Two methods use `var` parameters; these will be removed in Swift 3.

[image]

[image]

troystribling, 2016-03-05:
How do you turn on 3.0 deprecation warnings? I am not getting them.

cbowns, 2016-03-08:
If you build with Xcode 7.3 beta 4 or beta 5, it spits them out. https://developer.apple.com/xcode/download/

Bear in mind, some of the migrations in beta 5 (a.k.a. Swift 2.2 beta-ish?) are *not* backwards to Swift 2.1/Xcode 7.2.

troystribling, 2016-04-09:
Master now builds without warnings for Xcode 7.3.

closed 2016-04-09 by troystribling