troystribling/BlueCap #30: CC2650 SensorTag Support

rahimizad, 2016-04-11:
Hi Troy, I am unable to discover or connect to the sensors on the new CC2650 sensor tag. Is you TISensorTagServiceProfiles for the first gen or the current sensor tags? http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User's_Guide

troystribling, 2016-04-12:
I only have defined GAT profiles for the older version. I just ordered a new version. It does not support the new services. You should still be able to connect to it and discover services but they will show up as Unknown. What exactly do you mean by ' discover or connect to the sensors'.

rahimizad, 2016-04-12:
Hi Troy, the examples didnt work since they were looking for the accelerometer at the old UUID address. No worries, now that i know, I can update them to match the new sensortag. thanks so much

troystribling, 2016-04-12:
The examples will not work with sensor tag since it does not advertise services. The accelerometer example expects the accelerometer service to me advertised. If you build the Bluecoat app it should see the device.

mph070770, 2017-04-27:
Hi Troy. Sorry for opening an old thread. Can the CentralManager demo be fixed to work with the sensorTag? How do you connect to the sensorTag UUID of F0000000-0451-4000-B000-000000000000 instead of the accelerometer UUID F000AA10-0451-4000-B000-000000000000.

troystribling, 2017-04-27:
I do not understand the question.  F0000000-0451-4000-B000-000000000000 is the TI base UUID it does not map to a service. http://processors.wiki.ti.com/index.php/SensorTag_User_Guide.

If you want to add a service you can use this https://github.com/troystribling/BlueCap/blob/master/BlueCapKit/Service%20Profile%20Definitions/TiSensorTagProfiles.swift as a guid. 

The documentation on writing GATT profiles is here https://github.com/troystribling/BlueCap/blob/master/Documentation/GATTProfileDefinition.md.

closed 2017-06-11 by troystribling