troystribling/BlueCap #34: Crash when using BCCentralManager during application launch.

joewhaley, 2016-06-28:
I ran into a repeatable crash when I open my app via a push notification, and the BCCentralManager is started in didFinishLaunchingWithOptions. The Optional BCCentralManager.cbCentralManager is nil during the didUpdateState() callback. The problem is there is a race condition because the constructor sets the delegate to "self" before assigning to self.cbCentralManager, so if you get an event (like PoweredOn) during initialization, the didUpdateState() callback will happen before self.cbCentralManager is set, and you will get a crash.

I was able to fix it by passing in nil to the CBCentralManager delegate constructor, then explicitly setting the delegate after initialization:

E.g.
`
    public override init() {
        self.centralQueue = Queue("us.gnos.blueCap.central-manager.main")
        super.init()
        self.cbCentralManager = CBCentralManager(delegate: nil, queue: self.centralQueue.queue)
        self.poweredOn = self.cbCentralManager.state == .PoweredOn
        self.startObserving()
        self.cbCentralManager.delegate = self
    }
`
An alternative way to fix it would be to check the Optional cbCentralManager before using it.

troystribling, 2016-06-29:
I have seen this crash. How did you reproduce it?

joewhaley, 2016-06-29:
I was able to consistently reproduce it by waking up the app with a push notification via APNS.

troystribling, 2016-07-03:
I decided to us the centralManager passed to the delegate in didUpdateState. It should also resolve issue. I also made similar change to PeripheralManager.

closed 2016-07-03 by troystribling