troystribling/BlueCap #39: Crash sometime when rescanning  device !

duongsavvycom, 2016-09-22:
SimpleFutures.swift line 439
Future.complete(Try<A>) -> ()

duongsavvycom, 2016-10-10:
here is the log 
#0. Crashed: com.apple.main-thread
0  libswiftCore.dylib             0x1561d70 specialized _fatalErrorMessage(StaticString, StaticString, StaticString, UInt) -> () + 72
1  libswiftCore.dylib             0x160e5c5 _swift_stdlib_reportFatalError + 46
2  libswiftCore.dylib             0x142e648 _IgnorePointer.init() -> _IgnorePointer<A> + 1638
3  libswiftCore.dylib             0x14438b8 Array._copyToNewBuffer(Int) -> () + 776
4  BlueCapKit                     0x8529f8 FutureStream.addFuture(Future<A>) -> () (SimpleFutures.swift)
5  BlueCapKit                     0x852410 FutureStream.complete(Try<A>) -> () (SimpleFutures.swift:860)
6  BlueCapKit                     0x85280c FutureStream.failure(NSError) -> () (SimpleFutures.swift)
7  BlueCapKit                     0x851fdc StreamPromise.failure(NSError) -> () (SimpleFutures.swift)
8  BlueCapKit                     0x7e57dc specialized BCPeripheral.shouldFailOrGiveUp(NSError) -> () (BCPeripheral.swift:733)
9  BlueCapKit                     0x7e6a88 specialized BCPeripheral.didDisconnectPeripheral(NSError?) -> () (BCPeripheral.swift)
10 BlueCapKit                     0x7dc24c BCPeripheral.disconnect() -> () (BCPeripheral.swift)

troystribling, 2016-10-11:
How do you reproduce this error?

duongsavvycom, 2016-10-31:
We have a crash here : 
Crashed: com.apple.main-thread
0  BlueCapKit                     0x10085094c BCPeripheral.(discoverPeripheralServices([CBUUID]?, timeout : Double) -> Future<BCPeripheral>).(closure #1) (BCPeripheral.swift)
1  BlueCapKit                     0x1008b245c FutureStream.(onSuccess(ExecutionContext, success : (A) -> ()) -> ()).(closure #1) (SimpleFutures.swift:925)
2  BlueCapKit                     0x1008b0638 Future.(onComplete(ExecutionContext, complete : (Try<A>) -> ()) -> ()).(closure #1).(closure #1) (SimpleFutures.swift:484)
3  libdispatch.dylib              0x183b05200 _dispatch_call_block_and_release + 24
4  libdispatch.dylib              0x183b051c0 _dispatch_client_callout + 16
5  libdispatch.dylib              0x183b09d6c _dispatch_main_queue_callback_4CF + 1000
6  CoreFoundation                 0x184c29f2c __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 12
7  CoreFoundation                 0x184c27b18 __CFRunLoopRun + 1660
8  CoreFoundation                 0x184b56048 CFRunLoopRunSpecific + 444
9  GraphicsServices               0x1865d9198 GSEventRunModal + 180
10 UIKit                          0x18ab30628 -[UIApplication _run] + 684
11 UIKit                          0x18ab2b360 UIApplicationMain + 208
12 TurnPoint                      0x10014f4ec main (AppDelegate.swift:14)
13 libdispatch.dylib              0x183b385b8 (Missing)

duongsavvycom, 2016-10-31:
[image]

troystribling, 2016-12-22:
The newest release 0.3.0 has many fixes related to threading which could have caused this crash.. If you continue to see problem please open a new issue.

Thanks

closed 2016-12-22 by troystribling