troystribling/BlueCap #43: next_release example

cesargut, 2016-11-05:
Hi, I'm trying to test next_release branch but I'm new in Swift, what's the new syntaxis to create a new connection and when it's successful, it could probably be something like this? Could you give me a more complete example please?

manager.startScanning(forServiceUUIDs: [CBUUID]?).flatMap(mapping: (Peripheral) throws -> Future<M>)

Thanks in advance

troystribling, 2016-11-06:
I should be finished in a couple of weeks and will update documentation. Something like this should work, You need to define the errors thrown below they are not part of the framework,

```swift
connectionFututure = manager.whenStateChanges().onSuccess { state in
    switch state {
    case .poweredOn:
       return manager.startScanning(forServiceUUIDs: [CBUUID]?)
    case .poweredOff, .unknown, .unauthorized, .resetting, .unsupported:
       throw invalidCentralManagerState
      }
 }.flatMap { peripheral in
     peripheral.connect()
}

connectionFuture.onSuccess { [weak self] (peripheral, connectionEvent) in
    switch connectionEvent {
     case .connect:
     case .timeout:
     case .disconnect:
     case .forceDisconnect:
     case .giveUp:
      }
}

connectionFuture.onFailure { error in
}
```

closed 2016-11-06 by troystribling

cesargut, 2016-11-07:
Thanks @troystribling, but your example didn't work

When I add this:
```
let connectionFututure = manager.whenStateChanges().onSuccess { state in
  switch state {
  case .poweredOn:
    return manager.startScanning(forServiceUUIDs: [serviceUUID])
  case .poweredOff, .unknown, .unauthorized, .resetting, .unsupported:
    throw invalidCentralManagerState
  }
  }.flatMap { peripheral in
    peripheral.connect()
}
```
I get an error: Unexpected non-void return value in void function

When I remove "return", I get this error:
Value of tuple type '()' has no member 'flatMap'

The way how I installed it was like this:
pod 'BlueCapKit', :git => 'https://github.com/troystribling/BlueCap.git', :branch => 'next_release'

troystribling, 2016-11-07:
You need to add more type information. Sometimes the compiler cannot do
type inference on flatMap because it is overloaded.

cesargut, 2016-11-07:
I have no idea what you are talking about, the example you sent me is useless :(
I guess I'll have to wait until you release the documentation.

troystribling, 2016-11-07:
Add return types to closures.

troystribling, 2016-11-14:
Her is an example https://github.com/troystribling/BlueCap/blob/next_release/Examples/Central/Central/ViewController.swift#L118-L183.