troystribling/BlueCap #44: CBCentralManagerInjectable protocol conformance error

rameswarios, 2016-11-15:
I'm developing my app in Swift 3.0 and added BlueCapKit using CocoaPods. 
I've updated the flag '**Use Legacy Swift Language Version**' to '**Yes**' for BlueCap.

However I'm getting this particular issue which says  **Type 'CBCentralManager' does not conform to protocol 'CBCentralManagerInjectable'**

And this happens due to the **'state'** variable.

Here's a screenshot of the error to help you. Can you please help solve this?

[image]

troystribling, 2016-11-16:
This is fixed on the next release branch. Current master will not work with iOS  10 because of a change in the CBCentralManager and CBPeripheralManager state variable.

rameswar54, 2016-11-16:
Is **next_release** branch stable to integrate in project for Swift 3.0 version are need to wait?
Are you going to release in near future?

troystribling, 2016-11-16:
I plan to release soon. The interface is stable and no more changes are
planned. I am just working through examples and fixing bugs.

rameswarios, 2016-12-20:
Hello Troy, did you fix all the bugs?

Can I update my project which uses BlueCap already? I'm just holding on for BlueCap Swift 3 version. Or you're on the process still to fix all bugs and release soon. Any feedback shall be greatly helpful.

troystribling, 2016-12-20:
This should be resolved with version 0.2.0.

closed 2016-12-20 by troystribling