troystribling/BlueCap #51: tvOS Support

Dschee, 2017-01-19:
First thanks for this great library and example app!

As far as I can tell this doesn't support tvOS though (Carthage couldn't find a scheme for tvOS) which is a pity. Do you plan to add tvOS support and do you think there are parts that won't work or make any sense on tvOS? If so, which ones are they?

Thank you for your help!

Dschee, 2017-01-19:
I've just implemented this feature via #52 – here's the answers I found myself:

- the location features (beacon etc.) are not supported by tvOS
- some peripheral functionality (especially used in initializers) isn't supported in tvOS
  - see also http://stackoverflow.com/questions/33476994/defining-custom-services-and-characteristics-on-tvos

troystribling, 2017-01-20:
Thanks for doing this. I will review the PR and verify that it does not break any existing functionality and verify the build in a simulator but do not currently have the hardware to test tvOS BLE support. If it looks good I will do the merge.

closed 2017-01-21 by troystribling