troystribling/BlueCap #53: background

marwan87, 2017-01-19:
when i try and enable background it doesnt seem to be working, all the permissions are there and the code is going through and verifying everything in the library but the monitoring doesnt seem to be picking up the beacons.

if swap to events ranging that works perfectly, when i swap to All or region boundary the enter / Exit in onStateDidChange is not being called. 

and in background, when the app is not in the foreground everything stops straight away. 

Im i doing something worng

troystribling, 2017-01-20:
Are you using the BlueCap example? If so Central mode is not programmed to work in the background but Beacon ranging and Peripheral mode should work in background. 

Central applications written using the framework should work in the background if programmed to do so and the appropriate permissions are given.

marwan87, 2017-02-13:
Hi @troystribling sorry for the late reply i managed to get that part of the project to work, thank you for the advice. i also found where my Bug was.

The second part of the project im using the BlueCap example and im connecting to a sensorTag, what i would like is to still be listening to the button clicks when the app is in the background, is that possible?, or is that not accounted for in the library? 
I have got it all to work in foreground, thank you so much for the awesome library.

Thanks again for the library and taking your time

troystribling, 2017-02-13:
Nothing in the framework will prevent you from running in the background if the correct permissions are requested. See https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html.

From experience your time to do something is limited to about 15 seconds after receiving a bluetooth event. After 15 seconds your app will be suspended. You can start a background task and extend the time to 3 minutes.

marwan87, 2017-02-14:
15 seconds should be more than enough, thanks for your time, i will see how i get on with that, but background bluetooth is always a pain on IOS

closed 2017-04-28 by troystribling