troystribling/BlueCap #56: Scanning services automatically filter out same services?

soonfay, 2017-02-02:
Hi, the device we create has 2 same UUID services for battery from different sources. When I use bluecap service discovery, it automatically filter out the service. So when I use other BLE app, I will able to find lets say total 5 services, but bluecap only show 4 services due to same UUID? Is it possible to remove the filter option? If so , how can I do it? 

Thank you.

troystribling, 2017-02-02:
This would be a big change to the way services are identified and accessed by the framework. Currently the UUID is used as hash key. You would have to use CBService references to identify objects. The same would go for Characteristics.

This may be considered for a future release. I would also take a PR if  is something you want to do.

troystribling, 2017-02-03:
I looked into this on the branch https://github.com/troystribling/BlueCap/blob/multi-uuid/BlueCapKit/Errors.swift. Try it if you want. Pulling the change into the project would be a big effort. I think the that the request is legitimate but I would like to see if there is more interest.

soonfay, 2017-02-03:
Thank you for the response. I downloaded it and I will try to see if I can do it later since it most likely gonna take a while. Right now I have to other focus that I need to pay attention on.

soonfay, 2017-02-03:
Even it is not directly related, I found out that the number of services that being discovered for the first time is cached. I have a scenario  where new service will only showed with certain command and IOS will never see that unless I reset the phone. And the service will never disappear once it showed up even the operation is done. Is there a way to force clearing the cache of services to treat it as first discovery?

troystribling, 2017-02-03:
The cache is cleared every time you run discovery.

soonfay, 2017-02-03:
Tha'ts what I thought too but It didn't find the new service even when I tried with bluecap, the number of services stay the same. Android can find the new service but not IOS.

troystribling, 2017-02-03:
I misunderstood the question. The framework cache is cleared every time you
run discovery. The iOS BTService cache is not. I have seen this if I change
the GATT profile on a device. You continue to see the old GATT profile
until the iOS BTService cache is cleared. There is no way to clear the
cache from an application. Toggling Bluetooth off and on can clear it. I do
not know a way around your issue other than to not change the GATT profile
after first discovery.

soonfay, 2017-02-03:
Thanks for the quick response, I just tried to turn off and turn on the bluetooth manually and it is able to find the new service. Too bad it's part of the design for profile changes for indicate for any update and IOS don't allow turning off and on the bluetooth programmatically. Didn't expect this to be a problem.

troystribling, 2017-02-04:
The BlueCap app should now work for duplicate service and characteristics UUIDs. I also updated the PeripheralManager to support duplicates. I did a test with the peripheral simulator in the app and it worked. I do not have any hardware to test against. When you get a chance give it a try and let me know how it goes.

soonfay, 2017-02-06:
How should I add this new framework? I tried to remove the previous bluecap and add this one as new dependency but I get errors from compiling. Is it possible to install using cocoapods?

troystribling, 2017-02-06:
You can reference the framework pod with a path in your pod file. I wold do a `pod deintegrate`, `pod clean` and `pod Install`.

In you pod file add this to your target,

```
target 'MyApp' do
  pod 'BlueCapKit', :path => '../../'
end
```

With the path the location of `BlueCapKit.podspec`.

soonfay, 2017-02-07:
Hi,
I just tested the device with it, and it is displaying both battery level with same UUID! It is great, I am going to integrate it. Thanks for the help.

closed 2017-02-07 by soonfay