troystribling/BlueCap #57: Question: Is it possible to connect to a device with just MacAddress without scanning?

soonfay, 2017-02-21:
Lets say I saved the mac address locally, and I want to connect to the device without the scanning process. It will return either fail or success.

troystribling, 2017-02-21:
CoreBluetooth does not expose the MAC address so you cannot connect using it. Scanning is the only way to obtain a peripheral reference. After a peripheral is discovered you can retrieve if from the CoreBluetooth cache by either peripheral UUID or service UUID.

soonfay, 2017-02-23:
OK, I will save the UUID instead. I have problem with characteristic read. How can I read non-string value or in general? It say it's able to read successfully in readFuture but because the data is not string, it return nil. Even when I use BlueCap, the only value that I can read was battery level. 

Need some help, thanks.

troystribling, 2017-02-23:
You need to read to write a service profile like https://github.com/troystribling/BlueCap/blob/master/BlueCapKit/Service%20Profile%20Definitions/TiSensorTagProfiles.swift#L215-L306 or read the value as Data and cast the bytes to a C struct or covert them to swift objects. See https://github.com/troystribling/BlueCap/blob/master/Documentation/SerializationDeserialization.md for a discussion of how to deserialize data.

soonfay, 2017-02-27:
Hi troy,
Apparently it was the multi-uuid branch that wasn't able to read the value. It seem to be something with the changes of the characteristic reading compare to the main branch. I try to uninstall the multi-uuid branch BlueCap and install the main branch BlueCap, it was able to read the characteristic String values.

troystribling, 2017-02-28:
There was a bug causing the characteristic string value method to return nil if no characteristic profile was defined. It pushed a fix to paster and will release a new pod with version 0.5.1. Thanks for finding the bug.

soonfay, 2017-02-28:
Yeah sure, please let me know when its fixed.

troystribling, 2017-02-28:
It is fixed on the current master. I checked it  with a device I have. Would you mind trying?

soonfay, 2017-02-28:
There was a moment of confusion because main-branch still say pod 5.0 until i did pod update to 5.1 . Yes, its working with multi-uuid included. Thanks a lot.

closed 2017-02-28 by troystribling