troystribling/BlueCap #59: Resources are insufficient when writing characteristic

soonfay, 2017-03-06:
Hi Troy,

I have this issue everytime when I try to write a command. I cant really find anything helpful regarding to this issue. 

[image]

troystribling, 2017-03-06:
This is not a framework error. It is a CoreBluetooth error: https://developer.apple.com/reference/corebluetooth/cbatterror.code/1518710-insufficientresources

Here is some talk about it. Find9ing any information seems difficult.
http://stackoverflow.com/questions/40532822/meaning-of-insufficient-resources-in-core-bluetooth.

The discussions indicates the discussion indicates that it may be comping from the the device not the phone.

Writes work in examples and on hardware I have tested. Not sure I can do anything without the hardware.

soonfay, 2017-03-06:
Yeah, that's the only discussion I can find online. 
Ok, thanks for the quick response.

closed 2017-03-06 by soonfay