troystribling/BlueCap #64: Questions on MFI and Connectionless Slave Mode

marcosscriven, 2017-06-14:
I have some C code that communicates at a low level, using the [BlueZ lib](http://www.bluez.org/), with the Bluetooth HCI on the Raspberry Pi 3, allowing it to emulate a peripheral. 

I'm wondering if either my iPhone 6 or iPad Pro 10.5 has the capability to do this too. There's two parts to this:

1) It's still not clear to me if this would need MFI? So far as I can tell from Apple's docs, almost anything that works on Bluetooth LE is exempt, but I see contradictions to this in various posts.

2) Whether iOS and underlying hardware allows some of the Bluetooth modes I need

Looking at your excellent docs, I do see ```PeripheralManager``` - however, this is at a higher abstraction than I'm used to. For instance, I have this piece of code:

```
bt_hci_send(hci_dev, BT_HCI_CMD_SET_EVENT_MASK_PAGE2, ...);
```

Which sets the Raspberry Pi bluetooth module to something called "connectionless slave broadcast", setting it up to listen for ```BT_HCI_EVT_CONN_REQUEST``` from the master device, and if so commence with the synchronization train. 

I know not even all third party Bluetooth 4 chips support this mode, so wondering if iOS/BlueCap supports this? I can't see anything like it in your code.

troystribling, 2017-06-14:
My understanding is that Bluetooth LE is exempt from MFI. I have worked for companies that make BLE devices and MFI was not required.

If I understand your question correctly ```PeripheralManager``` would do what you are asking.  The BlueCap app, https://itunes.apple.com/us/app/bluecap/id931219725?mt=8#, can simulate TISensorTag services using it. And this example simulates a TISensorTag Accelerometer service, https://github.com/troystribling/BlueCap/tree/master/Examples/PeripheralManager.

marcosscriven, 2017-06-14:
Thanks @troystribling for the swift response. Will dig into the example you linked.

closed 2017-06-14 by marcosscriven