troystribling/BlueCap #65: CBMutableDescriptors can't be added to MutableCharacteristics

MrSage, 2017-06-14:
As far as I can tell there is no way to add a CDMutableDescriptor to BlueCap's MutableCharacteristic class. The underlying CBMutableCharacteristic is not exposed either so there is no way to add the descriptors directly to the characteristic itself.

troystribling, 2017-06-14:
```CDMutableDescriptor``` is not currently supported because I have one seen them implemented on a device.

MrSage, 2017-06-15:
I'm in the process of trying to implement the HID specification and they are required. Would it be possible to make the underlying characteristic visible as a short term workaround until a larger need for `CBMutableDescriptor`s arise?

troystribling, 2017-06-15:
I am sure not understand this question "Would it be possible to make the underlying characteristic visible?". 

Now the interfaces to access Descriptors are not exposed on the protocols that wrap the CoreBluetooth objects. You could add the methods to discover, access and add descriptors to the protocols to expose them. The CoreBluetooth wrapper protocols are here https://github.com/troystribling/BlueCap/blob/master/BlueCapKit/Injectables.swift. You could add the methods you need to the appropriate protocols. You could then write methods on BlueCapKit Peripheral and Characteristic that delegate to these methods on the CoreBluetooth objects.

MrSage, 2017-06-15:
I'm willing to take a stab at it, but I've only just figured out how to use this library. If you can give me some help I'm willing to put in the effort!

MrSage, 2017-06-16:
Hmmm... That was a lot easier than I was expecting. All I did was add a function that exposed `.descriptors` on the underlying CBCharacteristic. I imagine its not in line with the spirit of the project, but it seems to be sufficient for what I need. Swift is cool language...

troystribling, 2017-06-16:
To get it working so that you can use it is a small change. Adding it as a first level citizen is not much more work. If you want to keep using it we can discuss it further.

closed 2017-09-14 by troystribling