troystribling/BlueCap #77: Issue with connect inside custom framework

ankit2002, 2018-05-15:
Hi,
I am creating my own framework and using bluecap framework inside it. But I am getting a strange behaviour of app on first run. 
Scenario:-
1. Launch first build it found peripherals, services and characteristics and work well.
2. Now launch again it will found the peripheral and calls the connect method with timeoutInterval but never get any callback it just stuck there.

I checked the code of connect method and there is a reconnect method but don't understand why the timeout doesn't worked. In your test application it works well. Is it because I am using it in a custom framework? 
Thanks

troystribling, 2018-05-16:
I do not know. I have never used in a framework but do not think would cause any problems. Do you have a simple example?

ankit2002, 2018-05-16:
Thanks..... I am done with It. Some how its delegate were not getting called to disconnect after timeout. Thanks a lot 👍

closed 2018-05-16 by ankit2002