troystribling/BlueCap #78: Not getting callback for peripheral Disconnection

ankit2002, 2018-05-25:
Hey,
If I disconnect the peripheral (bleno in terminal with control+c) i am not getting any callback in Bluecapkit framework. It only get notify on didModifyService of Peripheral.swift. Will you please guide me which class or method should I look in to get that delegate callback for disconnection.

Thanks

troystribling, 2018-05-25:
Did you try this https://github.com/troystribling/BlueCap/blob/master/Documentation/CentralManager.md#central_peripheral_connection

ankit2002, 2018-05-26:
Thanks a lot for your response. Yes I have tried and I think its not the frameworks problem.
Actually i am using bleno as peripheral and killing it by pressing control + c in terminal and then I am not getting any disconnect callback. But the framework does get peripherals didModifyServices callback. The problem is iphone doesn't disconnect the bluetooth connection from mac. Do you have any advice for the approach.

troystribling, 2018-05-26:
Sorry, I do not have any experience with bleno. If you turn off bluetooth on the computer running bleno do you get the disconnect event?

ankit2002, 2018-05-26:
Yeah, If I turn the bluetooth off from the mac it does notify. Thats why I don't think its an issue on framework. Thanks anyways have a nice weekend :).

closed 2018-05-26 by ankit2002