troystribling/BlueCap #79: Limit on number of characteristics notifying?

magdkudama, 2018-06-13:
Is there any known limit to the number of characteristics that can be notifying at a time using BlueCap ?

troystribling, 2018-06-13:
None other than any limits that exist in CoreBluetooth.

magdkudama, 2018-06-13:
Thanks for your reply. Do you know if CoreBluetooth has any limits? I know it's nothing to do then with BlueCap, so I'll just close the issue. But if you know of any limits in CoreBluetooth, I'd appreciate your help.

Currently notifying for 5 characteristics, and in Android it's working fine, but in iOS using BlueCap it's not (for one of them).

Thank you for your work on this library, the API is amazing!

closed 2018-06-13 by magdkudama

troystribling, 2018-06-13:
There are limits on the number of connected peripherals, a few, before CoreBluetooth will crash causing the CentralManager and PeripheralManager to reset. I have not tested this tough since iOS 10. I would suspect there is a limit. The only way to know is to try. Hitting the limit will likely cause CoreBluetooth crashes not an error.