troystribling/BlueCap #82: Connection and disconnection not working properly

JohnyKumar27, 2018-08-01:
Hi, Thanks for the awesome library.
Actually i am facing some issues with the connection.
I have created a Home screen and Device list in both screens i am trying to connect to peripheral.
when i connect on device list it gets connected and i can write on characterstic and manually disconnect after that.(connection onSuccess and onFailure getting called)
When i go back to home screen and and make connection with same peripheral then also it is getting connected  i can write on characterstic and manually disconnect after that.(connection onSuccess and onFailure getting called).
But when i second time on device list after calling .connect "onSuccess" and "onFailure" are not getting called.
App flow is Showing device list from home screen.

Please if you can help on this.

troystribling, 2018-08-01:
I do not know what App flow is. There could be any number of things wrong. I do not have enough information to help you.

JohnyKumar27, 2018-08-03:
Thanks for replying.
After disconnection with peripheral, When i am trying to connect again with same peripheral then its onSuccess and onFailure not getting called.
Is there any possibility that will happen?

troystribling, 2018-08-03:
Are you using the same Peripheral instance or rediscovering the Peripheral?  Did you originally disconnect? How many Peripherals are you connected to?  Do you have access to logs on the Device you are connecting? Can you see what it is doing?

JohnyKumar27, 2018-08-08:
No, I am scanning again and using that peripheral, But i checked there addresses are same. 
Yes i am calling disconnect before scanning again.
Connected with single peripheral
Yes i have access of log, i am connecting with Raspberry Pi device.
it is printing log of connect with peripheral even returning promise also after connection. but connection future completions(onSuccess,onFailure) are not getting called.

closed 2018-08-27 by troystribling