troystribling/BlueCap #85: BLE Central Manager doesn't timeout scan after successful connection

romansavrulin, 2019-03-14:
Bluecap from Carthage, version **0.7.0**.

The following simple code

```swift
let discoveryFuture = manager.startScanning(forServiceUUIDs: [CBUUID(string:uuid)], timeout:5.0).flatMap { peripheral -> FutureStream<Void> in
                log.info("scanFuture: Found peripheral UUID: \(peripheral.identifier.uuidString). Trying to connect")
                self.manager.stopScanning()
                self.peripheral = peripheral
            
                return peripheral.connect(connectionTimeout: 5, capacity: 5)
            }

```

will successfully timeout scan for not available service for the first time. But if I will scan for the service that is available and  connect to the device, the second scan attempt for unavailable service will never timeout.

Is it a Bluecap or Core Bluetooth issue?

troystribling, 2019-03-18:
That is the way it is implemented. You can stop the scan anytime after you connect.

romansavrulin, 2019-03-18:
@troystribling But the scan is stopped right after discovery. Why the second scan attempt doesn't employ timeout?

romansavrulin, 2019-03-18:
@troystribling Quick follow-up sequence 

1. disable the peripheral you are trying to discover
1. start scanning for it with timeout
1. you will get timeout as expected
1. enable the peripheral you are trying to discover
1. start scanning for it with timeout
1. you will successfully connect and stop scanning in the future
1. disconnect
1. disable the peripheral you are trying to discover
1. start scanning for it with timeout
1. you won't get timeout for that discovery attempt

Am I missing something?

troystribling, 2019-03-18:
Did not follow. That sounds like a bug. Start looking here. https://github.com/troystribling/BlueCap/blob/master/BlueCapKit/Central/CentralManager.swift#L212. The code is not that complicated.  I do not have the time available now to look into this but will be will ing to accept a PR.

romansavrulin, 2019-03-22:
@troystribling Ok, thank you. Will dive into this, once we will be working on new maintenance release of our App

romansavrulin, 2019-05-29:
Quick try to fix the timeout inside the library wasn't successful, so I've worked-around this by the following code

```swift
 func startDiscovery(uuid:String){

     scanTimeoutTimer = Timer.scheduledTimer(withTimeInterval: 10.0, repeats: false) { _ in
            log.info("Scan timeout has passed")
            if self.manager.isScanning {
                self.manager.stopScanning()
                //trigger timeout event here
            }
            self.scanTimeoutTimer?.invalidate()
        }
                
        let discoveryFuture = manager.startScanning(forServiceUUIDs: [CBUUID(string:uuid)]).flatMap { peripheral -> FutureStream<Void> in
                log.info("scanFuture: Found peripheral UUID: \(peripheral.identifier.uuidString). Trying to connect")
                self.manager.stopScanning()
                self.scanTimeoutTimer?.invalidate()
                return peripheral.connect(connectionTimeout: 10, capacity: 1)
            }
}
```